nefarius/ScpToolkit #264: The virtual driver for the xbox 360 controller is not detected as a controller (version 1.7)

DorCoMaNdO, 2016-04-10:
I updated to the new 1.7 update, then my controller wouldn't work. I tried to reinstall and it didn't work, I uninstalled the DS3 driver (as it was the one from the new updated, hoping it would use the older driver), didn't work, I tried to install an older version and it didn't work, I then saw that the driver for the virtual controller was updated too, I uninstalled that driver, reinstalled the older version yet again and then the controller was detected yet again.
The new virtual xbox 360 controller does appear, but it's not in the list of game controllers and does not receive/send any input.

charlesmilette, 2016-04-10:
Have you noticed that a clean wipe is needed?

DorCoMaNdO, 2016-04-10:
I did, and I performed it... The clean wipe does not seem to remove the older drivers once they're installed in the system.

charlesmilette, 2016-04-10:
Try to remove any Xbox 360 virtual controllers, PS3 controllers and the Virtual Bus from the Devices Manager and the Driver Store. Then retry installation of the drivers

DorCoMaNdO, 2016-04-10:
I've done that, uninstalling the driver was the only way to fix it. I've reinstalled the driver a bunch of times already and that did not help, the issue remains the same, the controller appears as a detected device in the devices manager, it recognizes it as an xbox 360 controller, but it won't work in games, won't send or receive any input, checking the "Game Controllers" setting doesn't show the controller there, while the older drivers do.

Neurotic0416, 2016-04-11:
Same problem here. Controller is connecting and got this error:
[image]

I see the connected controller in the scp monitor, but it doesn't working. I tried clean wipe and reinstall drivers but I got same result all the time.

nefarius, 2016-04-12:
That's because I'm an idiot >_< I somehow overlooked that the wrong hardware ID is in the `Ds3Controller.inf` file. If you have a look in your device manager you should see something like this:

[image]

For a manual fix, try replacing the assigned drivers manually:

[image]

[image]

[image]

[image]

[image]

Neurotic0416, 2016-04-12:
Thx. But when I installed the driver manually I got blue screen with "DRIVER_VERIFIER_IOMANAGER_VIOLATION (ScpVBus.sys)" message. I got this messege in 1.6 version of ScpToolkit before. Thats why I wan't to install 1.7

lilyesx6, 2016-04-15:
Thanks Nefarius it's really a an excellent software, I hope you ad a support feature for fake ps3 controllers. Keep the good work <3

Neurotic0416, 2016-04-18:
I used the Driver Verifier Manager, deleted existing settings and now I have not DRIVER_VERIFIER_IOMANAGER_VIOLATION error and the controller works fine.

nefarius, 2016-04-24:
I ran the latest `ScpVBus.sys` with the verifier enabled and didn't get an DRIVER_VERIFIER_IOMANAGER_VIOLATION error any more. Can you try again but first check which version of the bus driver you are running?

closed 2018-01-26 by nefarius