nefarius/ScpToolkit #265: The virtual controller isn't disconnected, even when "Always unplug virtual device" is checked.

charlesmilette, 2016-04-11:
On the latest beta, 64-bits Windows 10 build 14295.

[See the video here](https://www.youtube.com/watch?v=Vx2YfywCXgU&feature=youtu.be)

Just a minor bug, but I appreciate if fixed. Except that and the Xbox 360 Controller issue on this Insider build, everything works fine.

zombiezoom, 2016-04-12:
Lucky you, I couldn't even install the drivers because my DS3 and DS4 didn't show up in the list (the list was empty)! The CleanWipe utility almost f**ked up my system and I had to do a system restore. I am gonna wait for the stable release.

charlesmilette, 2016-04-12:
Instead of using the Clean Wipe utlilty, I've done it manually. So like that I'm sure nothing was missed and it has less chance breaking my system.

If they weren't on the list, it is problably because libUsbK is still installed on them.

MWeWe, 2016-04-12:
@charlesmilette can you tell us how to fully uninstall it manually

nefarius, 2016-04-12:
@charlesmilette I'm seeing ghosts; tested it so many times and it worked. Please try this:

 - Stop the service
 - Go to your installation directory
 - Delete the file `ScpControl.settings`
 - Start the service
 - Test again

Thanks.

nefarius, 2016-04-12:
A crafted a [guide on how to remove toolkit remains by hand](../wiki/Manual-driver-removal).

nefarius, 2016-04-12:
Oh c'mon you can't be serious:

[image]

I accidentally put the hardware ID of the Xbox 360 controller in the `Ds3Controller.inf` file... I have to package and release an emergency-build to fix that.

charlesmilette, 2016-04-12:
Removing the settings file didn't worked, neither did unchecking the box.

charlesmilette, 2016-04-12:
[ScpToolkit_Log-Package_Charles_1460480294.zip](https://github.com/nefarius/ScpToolkit/files/215540/ScpToolkit_Log-Package_Charles_1460480294.zip)

closed 2018-01-26 by nefarius