nefarius/ScpToolkit #269: DS4 works wired, pairs but doesn't respond on Bluetooth

Rezulux, 2016-04-16:
I've been wrestling with this a bit over the past few days, I used to use InputMapper but migrated over to SCPToolkit to use my DS3 in PCSX2 for pressure sensitivity, and decided to set my DS4 up in it as well, which resulted in a number of errors and uninstalls and reinstalls before I ended up using the Clean Wipe Utility, and now things seem to be working fine while my DS4 is wired, but when I pair it wirelessly through bluetooth, it no longer works properly.

I'm able to pair it fine, it connects, the light bar changes color depending on which slot I have it set to, and I'm able to set other things like light bar brightness and whatnot. But otherwise, the controller is entirely non-responsive, even when using the legacy monitor, when I check the input tester to make sure it's not just the programs I'm using not seeing the controller, it will see the inputs for a few seconds after pairing before it stops receiving input entirely. My DS3 works fine both wired and wirelessly, and the input tester shows input being received on either connection method, so I'm confused as to what could be the problem.

Jan200101, 2016-04-21:
Its a problem many users have and its easy to fix (if you now a how to get into the deep stuff) but I kept you from that trouble and made this pile of code yay.
hope it fixes your problem (which it did in most other cases)

[Your solution](https://onedrive.live.com/redir?resid=531AF8D5D12A8FC8!2755&authkey=!AEphx3qq60WTNHs&ithint=file%2czip)

Rezulux, 2016-04-21:
The zip file you linked seems to only contain a readme text file?

nefarius, 2016-04-24:
There is indeed just a README included.

Rezulux, 2016-05-01:
Not to prod but there appears to be no change in the zip file, nor has my issue been resolved. It is still only a minor inconvenience, but there's been several times I've picked up my controller ready to play something only for me to realize it's on and my computer sees it, but it's not actually working and have to plug it in.

The readme file does not fix my issue, it mentions a tool to fix problems with the service, but the zip does not contain said tool. I'm not afraid to get my hands dirty within the program, so if there's anything I can be pointed towards that may help to maintain connection between my controller and SCPToolkit, I'd be more than happy to dig through and see if anything helps!

Jan200101, 2016-05-13:
@Rezulux sorry for no respond but I forgot my password (again) but now I am back.
You are saying there is just a readme ? I should never use onedrive again -.- but dont worry I am working on a better bluetooth fix right now that actually checks on startup

KinanEldari, 2016-05-22:
I've upgraded to a DS4 myself and I initially had it all working, but after an uninstall and reinstall for reasons I won't bore you with, I can get my DS3 to work normally and the DS4 works over the wire as others have stated above. But beyond "Dualshock 4 detected" popping up from the tray, then nothing connects and the controller turns itself off. I've tried to copy my initial installation from reinstalling the entire program (with use of the supplied clean-wipe utility along the way), removing the initial windows drivers for both the controllers and making sure I had everything as clean an install as I could, to timing the installation of the drivers differently. No dice.

closed 2018-01-26 by nefarius