nefarius/ScpToolkit #271: Massive FPS Drops When Using PS3 Controller

killa4000, 2016-04-17:
Even having the PS3 controller plugged in without the service running causes lag. 
I have no idea what caused this.
[ScpToolkit_Log-Package_phant_1460829780.zip](https://github.com/nefarius/ScpToolkit/files/222478/ScpToolkit_Log-Package_phant_1460829780.zip)

psyke83, 2016-04-21:
It's a bug in recent preview builds of Windows 10 related to the Xbox 360 driver, not ScpToolkit. Until it's fixed properly, you can workaround the problem:

* Plug in the controller
* Open Device Manager and click on "Xbox 360 Controller Driver for Windows"
* Click Driver tab -> Update driver -> Browse my computer for driver software -> Let me pick from a list of device drivers on this computer -> Show all devices, and install driver version 2.1.0.1349 [13/08/2009].

killa4000, 2016-04-21:
Yep that fixed it. Thanks!

psyke83, 2016-04-21:
There should be a "Close issue" or "Close and comment" right below the box where you type your reply (that I'm typing in right now).

closed 2016-04-21 by killa4000