nefarius/ScpToolkit #272: can't start SCP DS3 Service automatically

MWeWe, 2016-04-17:
i haven't got any answer from 8 days so i'm posting it again
[ can't use my controller wirelessly #246 ](https://github.com/nefarius/ScpToolkit/issues/246)

still having the same problem
here my new files

[ScpToolkit_Log-Package_WeWe_1460203395.zip](https://github.com/nefarius/ScpToolkit/files/222482/ScpToolkit_Log-Package_WeWe_1460203395.zip)

i have tried the exe that @Jan200101 gave to me but still the same
now after reverting back to ScpServer V1.2.2.175
Ds3Service not starting and when i tray to start is manauly i got this message

the SCP DS3 Service on local computer started and then stopped some services stop automatically if they are not in use by other services or programs

so i tried what @charlesmilette say here

sc config Ds3Service start= delayed-auto

but with no sucsses also tried this video
[How to Fix ScpService Not Installing on SCP DS3](https://www.youtube.com/watch?v=exAG3N2UWnQ)

with no sucsses also tried to change recovery under service properties with
first & second & subsequent feilures to restart the service
but it sill the same

also tried to increase the default time-out value for the service control manager to 120 seconds
[Increasing the timeout as described here](https://support.microsoft.com/en-us/kb/922918)

Note:: when i open ScpServer.exe on Scarlet.Crush version it works and i can use all my controllers
[ScpServer.V1.2.2.175.Log.zip](https://github.com/nefarius/ScpToolkit/files/222483/ScpServer.V1.2.2.175.Log.zip)

acme420, 2016-04-19:
do you have scp virtual bus driver in your device manager under system devices? does it have an exclamation mark? you might be having the same issue as me.

MWeWe, 2016-04-20:
i have it **without** exclamation mark

nefarius, 2016-04-24:
No need for multiple issues; we're all getting notifications on new posts.

MWeWe, 2016-05-24:
can't find any fix till now

charlesmilette, 2016-05-24:
It generally is a performance problem. Too many things are trying to start at the same time, so it slowdown this service's startup enough for it to get killed by the service controller.

A SSD or a better CPU could help a lot and fix the issue. If that isn't possible, try removing not used programs from your computer or clean installing.

MWeWe, 2016-05-25:
i have a ssd drive and i5 processor it only happens after reverting back to ScpServer V1.2.2.175

Jan200101, 2016-06-07:
To fix this set the service to delayed autostart and to restart if it crashed or failed to start sadly we cannot recreate something like that in batch yet to automate the process

thewriteok, 2016-11-26:
Hi, your batch script did not work for me, i got the same problem and i think that the root of the issue was that uninstalling/reinstalling the ScpDriver without rebooting made a service conflict, because when you uninstall it, the Ds3Service will be scheduled for deletion.

To solve just this problem just:
- reboot
- unistall
- reboot
- install

Maybe it could be possible to check for pending service deletions when installing the driver.

Anyway thanks for you excellent/free software.

MWeWe, 2016-11-27:
thewriteok  +1
AND i can`t find any way to add my controllers that works on old V1.2.2.175 to your new one

MWeWe, 2016-11-30:
Sorry I'm Such An idiot
i have disabled some of those windows Services

```
COM+ Event System
Computer Browser
DHCP Client
DNS Client
Network Connections
Network Location Awareness
Remote Procedure Call (RPC)
Server
TCP/IP Netbios helper
Wireless Zero Configuration ( XP wireless configurations only)
WLAN AutoConfig
Workstation
Network Location Awareness
Remote Procedure Call (RPC)
HomeGroup Provider
Network Connected Devices Auto-Setup
Windows Event Log
```

and that was what causing the problem
I AM SORRY AGAIN

closed 2016-11-30 by MWeWe