nefarius/ScpToolkit #273: Controllers not Showing in Device Manager

acme420, 2016-04-18:
hi i'm having a problem maybe someone can help me with? i was using the old version of scp fine with my ps3 controllers on usb. i bought a bluetooth dongle and tried to activate it but it kept giving me bluescreen errors in the scpvbus.sys. so i download this new one(latest 1.7.x) was on (1.6.2 working fine minus BT) and run the removal tool and reboot then run the new installer and follow all the steps it tells me everything is installed correctly and i hit finish but it never initializes the server/service. i open the monitor and it doesn't show the controllers. but if i open the legacy server app it shows both controllers and i can make them rumble. but the controllers don't show in windows or to the scp apps. i've rebooted and reinstalled several times but it doesn't seem to start the service.

*update* ok so i played around with it some more and it seems everything will work fine, USB and BLUETOOTH, if i manually start the scp server. all the controllers show in the monitor and profile manger. all the buttons work and the rumble works. but...windows still doesn't see the controllers so i can't actually play any games or do anything with them really. can anyone help me figure out this last part?

acme420, 2016-04-18:
[system.txt](https://github.com/nefarius/ScpToolkit/files/223281/system.txt)

acme420, 2016-04-18:
any i've been playing with this over and over uninstalling, reinstalling, searching the googles...

i cam across a post that shows how to manually uninstall so i uninstall, reboot. reinstall and carefully watch. it would seem that it's not installing the virtual x360 controller driver. "Virtual Bus Driver"  the first time you run it there's a quick little error message popup saying it couldn't install the the driver. then if you do it again subsequent times it says the drivers is installed fine, but it's clearly not installed or working. and in device manager it is listing an error.

Windows cannot verify the digital signature for the drivers required for this device. A recent hardware or software change might have installed a file that is signed incorrectly or damaged, or that might be malicious software from an unknown source. (Code 52)

according to this it's not signed?  my time is correct on my windows. i'm going to reboot and double check my system timer.

acme420, 2016-04-18:
i've nailed the issue down to something to do with the ScpVBus.sys file. it shows that your certificate is valid...but for some reason windows thinks it's unsigned. any help?
[image]

nefarius, 2016-04-18:
@acme420 does your system use UEFI (SecureBoot)?

charlesmilette, 2016-04-18:
WIndows 7 can't Secure Boot.

nefarius, 2016-04-18:
@charlesmilette d'oh, right!

acme420, 2016-04-19:
nah , i disable all the security bits Nx flag etc from bios. i use ACPI if that helps.

psyke83, 2016-04-23:
So this seems loosely related to issue #266.

acme420, please check Event Viewer -> Applications and Service Logs -> Microsoft -> Windows -> CodeIntegrity -> Operational. Do you see event id 3084 or 3085 there?

nefarius & charlesmilette - perhaps on Windows 7, WHQL enforcement is enabled by default (since Secure Boot is not supported), whereas the restriction is more relaxed in later versions of Windows for systems with Secure Boot disabled/unsupported?

charlesmilette, 2016-04-23:
@psyke83 Why would that be so?
@acme420 Try running this in a admin command prompt: `bcdedit /set loadoptions DDISABLE_INTEGRITY_CHECKS`
DDISABLE is not a typo.

This should disable driver signature enforcement, and the driver should work.

nefarius, 2016-04-24:
Does the last v1.6.x release work on your system?

acme420, 2016-04-27:
been away for a while. the 1.6 was working but the bluetooth didn't. i installed 1.7 hoping to fix that, but broke it altogether. i was booting with F8 and disabling driver signature checks, but when my computer goes to sleep i have to reinstall the driver again sometimes. and i've been having to use the manual server as well as only usb because the bluetooth stopped working and even thought the driver is still installed and shows up in windows the program doesn't see it. but i assume it's all related. 

@psyke83 no 3084 or 3085
@charlesmilette does this work right away or will i need to reboot? i'll prolly have found out before you reply.

acme420, 2016-04-27:
ok @charlesmilette i did the command you said and it worked right away without rebooting. i removed the scpvbus, and ran the cleanwipe then i reinstalled. everything went fine. the controllers install and detected right... so i unplug them to test the bluetooth...that works...but when i plug them back to usb they don't work anymore and now only work on bluetooth lololololol

charlesmilette, 2016-04-27:
Try restarting. That command should work only after a restart. It does the same thing as reboot, press F8, and boot without driver signing. It just is permanent.

acme420, 2016-04-27:
oh no i said it worked. i was able to successfully install. i just have another issue now of my controllers not working on usb anymore once connected on bluetooth.

nefarius, 2016-04-27:
I just tripple-checked the certificates and cross-certificates used in my production release signing process and according to MSDN I used the correct cross-certificate and tools so for now I'm honestly out of ideas on how to pin down the signing issues :(

psyke83, 2016-04-27:
nefarius,

https://www.osronline.com/showthread.cfm?link=274519

> This error maybe caused by the algorithm you used to code signing is SHA-1. The 
use of SHA-1 as a hashing algorithm for signing purposes has been discouraged 
and is no longer a best practice, and  Microsoft does NOT support for SHA-1 
hashing algorithm from 1 Jan 2016. Microsoft recommends using the SHA-2 hashing 
algorithm instead.

nefarius, 2016-04-27:
@psyke83 I'd rotate in glee if that's the issue :+1: 

@acme420 could you please try this:
1. Install the latest 1.7 beta as usual.
1. After you finished the installation, manually update the driver on the `Scp Virtual Bus Driver` using this specific version: [ScpVBus_x64.zip](https://github.com/nefarius/ScpToolkit/files/239098/ScpVBus_x64.zip)
1. Test and report back ASAP :)

psyke83, 2016-04-27:
I tried with no success, sadly. Would you be able to share the exact command line you've used via signtool to sign the driver? Are you specifying the cross cert correctly?

See, e.g.: https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/b3a1db5d-d81b-4b9a-bd08-5b3b69b7e708/windows-cannot-verify-the-digital-signaturecode-52?forum=wdk

> Turns out I did have the right cross certificate but I was confused by the instructions.
On the web page http://msdn.microsoft.com/en-us/windows/hardware/gg487315.aspx it says to find "Find the Issuer and Thumbprint for this certificate. Then locate the corresponding entry for this CA in the list below"
The Issuer is Verisign but the Thumbprint does not match the "Root certificate thumbprint" of VeriSign Class 3 Public Primary Certification Authority – G5. Turns out that the thumbprint is thumbprint of the cross certificate, not the root certificate.
When I use the new cross certificate then signing, verification, and installation succeed.

nefarius, 2016-04-27:
I let Visual Studio do the signing (yes, shame on my laziness), it uses the following command line:

`"CN="WDKTestCert Benjamin,1309XX...." | BD19668454XXX..." "E=nefarius@dhmx.at, CN="Open Source Developer, Benjamin Höglinger-Stelzer", O=Open Source Developer, C=AT | 61197CF2XXX..." "C:\Program Files (x86)\Windows Kits\10\CrossCertificates\Certum Trusted Network CA.crt" /fd"sha256" `

Where `Certum Trusted Network CA.crt` I gathered from [here](https://msdn.microsoft.com/en-us/library/windows/hardware/dn170454(v=vs.85).aspx).

psyke83, 2016-04-27:
I've no idea about Windows 7, but maybe your cert is not complying with the new requirements for Windows 10? See: https://msdn.microsoft.com/en-us/library/windows/hardware/ff548231(v=vs.85).aspx

Take a look at the table and compare the requirements for x64 for Secure Boot enabled vs disabled.

If that's the problem, there may be a way around the requirement for an EV cert, etc.; read the "Windows 10 kernel modules must be signed by the portal" section: http://www.davidegrayson.com/signing/

nefarius, 2016-04-27:
This time I manually signed the catalog and the driver like:

`signtool sign /v /ac "C:\Program Files (x86)\Windows Kits\10\CrossCertificates\Certum Trusted Network CA.crt" /sha1 6119XXX... /fd sha256 /tr http://sha256timestamp.ws.symantec.com/sha256/timestamp/ scpvbus.cat`

`signtool sign /v /ac "C:\Program Files (x86)\Windows Kits\10\CrossCertificates\Certum Trusted Network CA.crt" /sha1 6119XXX... /fd sha256 /tr http://sha256timestamp.ws.symantec.com/sha256/timestamp/ scpvbus.sys`

Release for tests: [ScpVBus_x64.zip](https://github.com/nefarius/ScpToolkit/files/239283/ScpVBus_x64.zip)

acme420, 2016-04-27:
@nefarious i disable the nointegritychecks (turn signing back on) and i reboot, then i run clean wipe...(prolly should have rebooted a second time. will try again in a minute) then i run the installer. it goes through, i get the error message about the SCPvBus too quickly to even see cept i know what to look for. finish the installer and PS3 controllers in devices, but no Xbox360 controllers. i open folder with modified vbus and right click to install. error message says it doesn't support this method of install. i'm gonna try and move all the files to the right folders manually...

oh wait i see you put up another one...lemme try that real quick...

nefarius, 2016-04-27:
Nonononono, not install via the .inf file, you need to hit the Update button on the existing bus in Windows Device Manager.

acme420, 2016-04-27:
yeah same error, "this INF doesn't supposrt this method of installation" might just be my windows acting up...i noticed today that i can't open my group policy editor anymore (w7x64 ultimate). will investigate some more.

nefarius, 2016-04-27:
See my previous post.

acme420, 2016-04-27:
@nefarious oh lol i never install drivers like that. i usually just install from inf

nefarius, 2016-04-27:
@acme420 since the driver is for virtual "hardware" the procedure is a tad different.

acme420, 2016-04-27:
ok i tried to update with the second one and it says it's not signed. lemme check with the first updated one...yep same deal error 52 cannot verify signature.

acme420, 2016-04-27:
i'm gonna completely manually remove everything and start over again with a fresh reboot.

nefarius, 2016-04-27:
*scartches head*

nefarius, 2016-04-27:
Right.

nefarius, 2016-04-27:
Oh btw.: can you check if you have the following [Windows Updated](https://www.microsoft.com/en-us/download/details.aspx?id=46148) installed? In a pretty outdated test-machine of mine I had to install it for the signature check to work.

psyke83, 2016-04-27:
The manually-signed driver still has the same problem. I'm sorry to break it to you, but I think your certificate can't sign kernel-mode drivers on Windows 10 Secure Boot mode.

Again: http://www.davidegrayson.com/signing/#kmod

> Microsoft announced that all new Windows 10 kernel mode drivers must be submitted to and digitally signed by the Windows Hardware Developer Center Dashboard Portal, which is a web service provided by Microsoft. Also, 90 days after the release of Windows 10, the portal will only accept driver submissions from you if you sign them with an Extended Validation (EV) certificate, which is typically more expensive than a normal certificate.

>However, there is a really nice loophole. For backwards compatibility, kernel-mode drivers signed with a valid cross-certificate that pre-dates Windows 10 will continue to pass signing checks in Windows 10. The cross-certificate from GlobalSign was issued long before Windows 10 and it expires on 2021-04-15. Therefore, you should be able to sign kernel-mode drivers for Windows 10 with a regular GlobalSign code-signing certificate until then. I successfully tested this on a Windows 10 machine on 2015-11-10.

Your cross-certificate is dated 29/10/2015, after Windows 10 RTM. The older version of ScpVBus.sys that was signed by Bruce James (and works in Secure Boot mode) uses the GlobalSign cross cert issues on 13/04/2011.

Unfortunately, the GlobalSign certificate seems quite expensive compared to the one you're using.

nefarius, 2016-04-27:
[image]

acme420, 2016-04-27:
@nefarious i don't have that update installed. should i install it and try again?

nefarius, 2016-04-27:
@acme420 might be worth a try.

acme420, 2016-04-27:
i have good news and great news.
Good news we can call this resolved.
Great news is everything worked after installing the 3099 or whatever update up there, and i also installed another update that fixes windows update hanging. but one of those worked i didn't even need to manually update to the new scpvbus you sent. 

i unplugged the controller and it went to bluetooth then i plugged it back in but it didn't reconnect. i went to the settings manager and selected reserve mac address for all psx environment and i disabled the other option to delete the device when i unplug it. 

issues all solved.

acme420, 2016-04-27:
Oh yeah forgot to say THANKS FOR ALL YOUR HELP!! if you have a paypal i can donate you a couple of coins. (literally all i have lol)

closed 2016-04-27 by acme420

nefarius, 2016-04-27:
Uh cool, I guess...

[image]

Anyways, glad we could solve this. [Indeed I have a PayPal account.](http://forums.pcsx2.net/Thread-ScpToolkit-XInput-Wrapper-aka-ScpServer-Reloaded)

acme420, 2016-04-27:
lol i'm sure it was the sha-2 update that did it. i just updated my WUdater at the same time since it was giving me shit.

nefarius, 2016-04-30:
@psyke83 thank you very much for the [provided link](http://www.davidegrayson.com/signing/) it answered so many questions straight instead of the blurry outdated results many google searches gave me. I read about the loophole he mentions and consider acquiring a GlobalSign certificate if that fixes everything.

psyke83, 2016-05-01:
No problem, but I would urge you to double-check that information before buying a new cert.

I read the original announcement, and the wording is a bit confusing: https://blogs.msdn.microsoft.com/windows_hardware_certification/2015/04/01/driver-signing-changes-in-windows-10/

My first interpretation was that was that the date of the cross-certificate mattered (same as what David Grayson's blog says), but it's possible to interpret it differently. It may be the case that you can only continue signing code if *your* certificate (not the cross-cert) was issued before the Windows 10 release date. If that's the case, buying a new cert may not work. Take some time to read the comments in that link, as there's a lot of contradictory information.

handoman123, 2017-12-26:
if u all manged t install the driver as i did, but see the controller in devices n control panel, install kinoconsole, it adds a virtual xbox controller which u can keymap using UCR. And yes i`m using windows 10.