nefarius/ScpToolkit #274: Vibration isn't working at all on DS3 controllers

dynpan, 2016-04-18:
Hello

I was using the old (original) SCP package, completely normally without any issues, and then I switched over to this one. I'm not sure when exactly, but in the meantime vibration stopped working. I can't get it to work in the SCP application, in games, nothing, regardless of which controller I use (I have four). They work normally on the PS3 though. I tried uninstalling these drivers and doing the clean wipe, and then using the old package drivers and apps but it still doesn't vibrate at all. Where do I even start trying to fix this?

Thanks in advance

dynpan, 2016-04-18:
Debug file:

https://mega.nz/#!Ewgj1CaQ!TRjvh6ZQsFMfR8eoHQz43GXrhMSr5aMbRld8CvAImhU

(I have no idea how to use this, I just closed and stopped everything and ran this, do I have to play a little or something?)

InfiniteDerp, 2017-04-02:
Same issue here.

Remylenain, 2017-08-12:
Hi, 
Sorry to unearth this topic but i've got the same problem.

My DS4 working nicely, all buttons and vibration is OK.
But for my DS3 (one genuine and one fake), the vibration don't work at all.

closed 2018-01-26 by nefarius