nefarius/ScpToolkit #275: Random disconnects

swimmesberger, 2016-04-19:
I have random disconnects while playing using a USB-Cable and a genuine PS3-Controller. When the disconnect happen a unplug and replug of the USB-Cable resolves the issue most of the time but in some cases this doen't work and leads to a even stranger case that my bluetooth keyboard isn't working anymore. (even though I'm not using a bluetooth adapter for the controller or the bluetooth driver) The only way to fix this state is to restart the PC. After the restart the controller and the keyboard are working again. Can't really tell what leads to the problem sometimes a couple of seconds after starting playing the disconnect happens sometime it takes a couple of minutes - for me it seems pretty random.

acme420, 2016-04-20:
does it happen after a strong rumble?

swimmesberger, 2016-04-20:
I have disabled rumble because it lead to disconnects in the past this particular problem could be connected with the rumble problem but the controller doesn't really "rumble" when the controller disconnects (because it's disabled in the game). I'm on Windows 10 x64 btw.

closed 2018-01-26 by nefarius