nefarius/ScpToolkit #277: Left stick input without being touched

brlafreniere, 2016-04-22:
Using a DS3 controller, every so often, it appears the left stick is giving some kind of "phantom input", where it's as if the left stick is constantly being flicked to the left, even though it's not being touched.

If I disconnect the controller for a moment, and reconnect it, or if I push the little "reset button" on the back of the gamepad, it fixes it... momentarily. But then after a few minutes it starts again.

I've read about the dead-zone configuration, and I cranked both dead-zones all the way up, but this issue is still happening.

Not sure if this is a hardware issue or software issue.

nefarius, 2016-04-24:
Sounds like a hardware issue since the software simply translates what the controller sends in.

MinekeyTP, 2016-04-28:
Definitively seems to be a software issue, I think it's the same here: https://github.com/nefarius/ScpToolkit/issues/214

I was playing to black desert and I configured a shortcut on the left stick click and in game it was a nightmare. When the stick is on the click position + any direction several command are generated randomly (mostly dpad and L1/R1). I tried on two games with two different ps3 controller. Tried uninstall and clean install always the same issue.

Maybe i'm wrong but I think all those issues comes from the left stick click. It can appear random because this click isn't a commun button on games but people can press it by inattention. It's simple to test, just maintain the left stick click and roll the stick in any direction.

Mugunge, 2016-05-20:
Hi,

I have the same exact issue with my genuine DS3 (shows up as fake though). From time to time when I play games it'll just start spamming left-stick inputs to the left. I've kept XInput Tester open and I can see that the blue bars for both sticks randomly flicker very fast from time to time, even when the controller isn't in use. 

I uninstalled and went back to the 1.2.0.160 release from Scarlett and that has fixed the issue.

brlafreniere, 2016-05-21:
That is good to know, I'll try that version and see if it works, thanks!

KinanEldari, 2016-05-22:
this sounds very much like my issue. https://github.com/nefarius/ScpToolkit/issues/179 So far what I suspect may have fixed it for me is adding the first bit of my mac address of my genuine controller to the scpcontrol.ini file. That's the gist of it anyway. You can read the rest of the details in that post. Regardless, ever since I have not had this particular issue with the left stick.

RogueLizard, 2016-09-23:
Can confirm. Not only is it with SCP but it also happens with DS3Tool drivers. I think it has something to do with Windows and the Sony controller interface itself, not the controller or the driver. Using a genuine Sony controller with Ds3Tool (because SCP still sucks right now)

Eligament, 2017-03-30:
I'm really sorry for bumping a near-year old post, but I'm having this exact same issue, like litterly the exact same thing. I've taken my controller apart twice now to try and look for issues but seem to have found none. I would just like to know where I can find this 1.2.0.160 release from Scarlett at. Thanks.

RogueLizard, 2017-03-30:
My left stick has been drifting on like 4 different controllers, PS3, PS4,
doesn't matter. DS3Tool didn't have this problem. The deadzones don't do
shit.

closed 2018-01-26 by nefarius