nefarius/ScpToolkit #280: SCP DSx Servive Could not be installed

Setiadii, 2016-04-23:
Why when im install this aplication, showed massage like this?
What happen with this aplication?
Can you help me?
Thank you.

[image]

nefarius, 2016-04-24:
UAC is enabled? Still happening after reboot?

Setiadii, 2016-04-24:
Yes. It still happening after im reboot my notebook.
And it happened after im re-download this app to.
What happen about this app?
Please help me.
Because im very love this app.
Thankyou

Setiadii, 2016-04-24:
Can you fix this problem?

nefarius, 2016-04-24:
Have you tried to explicitly run the setup as Administrator?

Setiadii, 2016-04-25:
Yes.
I have do that.
But it still not work.

[image]

[image]

What should I do?

closed 2018-01-26 by nefarius