nefarius/ScpToolkit #281: USB extension cord not allowing controller to be recognized

zj44, 2016-04-23:
Hi,

I used an older version (pre-Nefarius) of the DS3 tool and it would work fine when I had my PS3 controller connected through my USB extension cord. I reinstalled my computer recently and while it does appear in my device manager when I have my PS3 controller connected with the extension cable, ScpToolkit doesn't acknowledge it. I could use it by plugging it directly into my computer but I'd like to use the controller across the room and the extension cable is the right length for it.

Thanks

Techman-, 2016-04-23:
If your issue is solved, go ahead and mark it as closed.

zj44, 2016-04-24:
I wanted to leave it open in case extension cords with PS3 controllers not working in USB 3.0 ports are something that need to be looked into

nefarius, 2016-04-24:
Nothing I could change about this since propper connectivity has to be assured by the operating system.

closed 2016-04-24 by nefarius