nefarius/ScpToolkit #282: Can't use DS4 with PC Remote Play.

Mahir007, 2016-04-24:
Recently I installed the latest stable build of SCP Toolkit (the on released in January), did the default install options, installed just the DS4 driver and the windows service (didn't install the bluetooth driver), worked perfectly with no problems whatsoever, that was until recently when I installed the Remote Play client that Sony released recently, it just doesn't work with remote play, even if I completely stop the SCP DSx Service with the command prompt that doesn't make a difference.
Anyways that I can keep the SCP driver installed and use remote play at the same time, so that when I'm playing PC games I can use the DS4 as normal and when I want to use Remote play I can do that as well without having to uninstall the driver.

charlesmilette, 2016-04-24:
#260

nefarius, 2016-04-25:
There is no way around manually changing the device driver if you want to accomplish this.

closed 2016-04-25 by nefarius