nefarius/ScpToolkit #284: Dell Built-in Bluetooth

remivamp, 2016-04-25:
This is the first time ive ever had bluetooth on a computer that the company actually PUT THE BLUETOOTH IN! (looking at you asus) Anyway I'm trying to connect my Genuine PS3 controller to my computer via bluetooth since usb is really finicky. Anyway the controller works fine over usb on my windows 10 Dell Inspiron 14-3452 in order to play minecraft properly. But the bluetooth Dell Wireless 1801 Bluetooth 4.0 LE when I override the drivers shows up as Bluetooth Radio and under usb devices yet I can still see its mac address on scp monitor

nefarius, 2016-04-26:
Which software version are you running?

remivamp, 2016-04-26:
first tried latest stable. Wiped then tried lated prerelease. Then tried the one from scarlet crush. BT works fine on everything except scptoolkit.

remivamp, 2016-04-26:
By works fine i mean when its using oem drivers

charlesmilette, 2016-04-26:
Normal. They were _conceived_ for that. The Scp Bluetooth driver is generic. Thus, it cannot take in account all existing hardware only like `*poof*`. Some changes in the code is probably required.

mirh, 2016-07-16:
#380

closed 2018-01-26 by nefarius