nefarius/ScpToolkit #285: Can't get Bluetooth to work (USB works just fine)

redeven, 2016-04-26:
Hey! I've been unable to get my PS3 controller to work wirelessly so far.
According to the original XInput Wrapper thread, my Bluetooth dongle should be compatible (USB\VID_0A12&PID_0001) and according to the github page, my controller is compatible too (USB\VID_054C&PID_0268) but the program says it's a fake DS3 controller.
I was able to make it work while USB-connected, but I can't use it with Bluetooth. 
Whenever I attempt to connect wirelessly, it'll just show these notifications https://imgur.com/NwTRR4r

machstem, 2016-05-05:
Did you end up getting it to work?  I have a similar issue where it detects it as compatible but only works by USB

redeven, 2016-05-05:
Not at all. There was an unofficial ScpControl.dll that someone made, for a beta version that I couldn't get it to work, and I got no other answers, so... that.

mirh, 2016-07-16:
Similar symptoms as #299?

closed 2018-01-26 by nefarius