nefarius/ScpToolkit #286: DS3 controller is being picked up but..

dragoicath, 2016-04-26:
while the input tester picks everything up it isnt being treated as an xbox 360 controller and wont work on any games
[image]

acme420, 2016-04-27:
same issue as me, the xbox drivers (scpvirtual bus)didn't install. the temporary fix is to reboot, press F8, and boot without driver signing then reinstall the drivers. you'll have to do this every reboot.

charlesmilette, 2016-04-27:
Or in admin cmd:
`bcdedit /set loadoptions DDISABLE_INTEGRITY_CHECKS`
DDISABLE is not a typo.

dragoicath, 2016-04-27:
@charlesmilette do i need to try to reinstall the drivers after doing that?

charlesmilette, 2016-04-27:
That's the best part of it: Nope! Just reboot.  
It will only work if the solution proposed by @acme420 worked. It is just a permanent way to apply it, instead than having to do his solution at each reboot. So I suggest you try @acme420's solution before, and if it worked, proceed with the command.  
If not, I can't assist you any furter than: try reinstalling the driver after a [manual removal](https://github.com/nefarius/ScpToolkit/wiki/Manual-driver-removal)

dragoicath, 2016-04-27:
welp ive completely removed everything and tried to reinstall, no change if anyone else has any ideas theyd be appriciated

acme420, 2016-04-27:
try installing this update and reboot. if you don't already have it installed. it worked for my issues of drivers not being recognized as signed. https://www.microsoft.com/en-us/download/details.aspx?id=46148

closed 2018-01-26 by nefarius