nefarius/ScpToolkit #287: "Genuine DualShock 3 found" but wouldn't connect

firezz1, 2016-04-27:
hi guys!
I'm having some issues regarding the bluetooth connection for dualshock 3.
I've installed the SCPtoolkit and the drivers for dualshock 3, xbox, and bluetooth. Everything is working perfectly fine, but then when i restart or wake up a computer from sleep the bluetooth wont work anymore, when i press the ps button there's a notification saying "Genuine Dualshock 3 found" and the light just keep blinking..
What might be causing this problem?

acme420, 2016-04-27:
open the settings manager and see if under service settings you have always unplug virtual controller checked. if so uncheck it and unplug/plug your controller on usb again. then try and unplug it and see if it works. if not try turning off the controller first. hold l1+r1 and ps button till it goes off then unplug/plug it back in and unplug it again to see if it works on bluetooth now.

also might want to try reserving the slot in the service settings last option.

firezz1, 2016-04-28:
i've checked the setting and the "always virtual controller" is unticked, for the reserving slot i already tried that and the result is still the same..
but!
yesterday i found something interesting, when i press the ps button just after it turned off because idle i found this

[image]

did you know what that means?

closed 2018-01-26 by nefarius