nefarius/ScpToolkit #288: Installed this now I cannot log into my pc

broodilicious, 2016-04-27:
I tried to install this and as I was installing the drivers my PC locked up I tried to restart after about 20 minutes of nothing happening and now my computer hangs on the login screen, I was able to access my PC through my backup HD but nothing I have tried including deleting the program seems to fix it

Zero3K, 2016-05-02:
That's really weird. What are the specs of your PC, the Bluetooth Dongle you used and where was it plugged into?

Jan200101, 2016-05-13:
Thats why I always say : Fingers away from a program if you cant save yourself from it

ninmonkey, 2016-05-14:
@broodilicious , @Zero3K 

I've had the exact same problem. After choosing which of the ~10 drivers to install for ps3, it froze up. No response to mouse or keyboard. I left it a while, then when I came back I rebooted it.

Windows reboots to the login screen. Mouse and keyboard are both plugged into USB. Neither mouse nor keyboard responds. 

**my fix to revert**: I plugged in a second USB keyboard. This one let me login, and I chose **start menu -> restart (while holding shift)**. That brought me to the windows recovery screen troubleshooting screen, which **let me revert to the recovery point** before SCP.

> That's really weird. What are the specs of your PC, the Bluetooth Dongle you used and where was it plugged into

No dongle so ps3 is using USB. PC is win10. I know I've used an older version of xinput/scp in the past, which worked. I grabbed the latest to install here.

edit: The link on this github to http://forums.pcsx2.net/Thread-ScpToolkit-XInput-Wrapper-aka-ScpServer-Reloaded has been down for multiple days.

charlesmilette, 2016-05-14:
> The link on this github to http://forums.pcsx2.net/Thread-ScpToolkit-XInput-Wrapper-aka-ScpServer-Reloaded has been down for multiple days.

We can't do anything about the PCSX2 forums being down. We have to wait.

> I think the problem Is I chose all the check boxes for ps3 on this screen:
image replacing my keyboard and mouse drivers?

That's the worst idea ever! This list is built with __all the USB devices plugged in your computer__. Selecting them all means that all of them will be overwritten with the libUsbK driver (read as: Windows won't know what the f*ck are those devices, and Scp assumes they are PS3 controllers). But Scp can only understand the PS3 controller input, not your mouse or keyboard's.

charlesmilette, 2016-05-15:
A solution would be to use [this guide](https://github.com/nefarius/ScpToolkit/wiki/Manual-driver-removal). Only the part named "Remove device drivers" concerns you. You can ignore the other parts if you plan to still use ScpToolkit.

Don't forget to stop the ScpToolkit standalone server or service before.
To stop the service, run `sc stop Ds3Service` in an elevated command prompt
To stop the standalon server, simply close it.

closed 2018-01-26 by nefarius