nefarius/ScpToolkit #289: Does not detect bluetooth dongle

blubbsy, 2016-04-27:
Hi,

after some time i updated now to the newest version of SCP and found out, that it does not work anymore with my dongle. Already with the old verisons i had to manually add it to the *.inf for the drivers to work properly, but then it worked without any problems :)

Maybe you can add the hardware id to your tool? :) The hardware ID is this one: USB\VID_0A5C&PID_2123

Thanks
Blubbsy

igorrafael, 2016-05-15:
Similar issue here: before nefarius took the project my built-in Bluetooth device didn't work because it was not compatible with the Bluetooth library used by SCP. When nefarius took over the project and changed it to one that did support my system so it worked fine until the latest beta. In the beta, conveniently named "Back-to-the-roots-Edition", my device is also not recognized.

I've reinstalled the current release and it still works the way I'm used to - needs a Bluetooth driver reinstall after every reboot but otherwise works fine.

mirh, 2016-07-16:
#380

closed 2018-01-26 by nefarius