nefarius/ScpToolkit #292: Unable to select and edit different sections in the Profiler

Encryption767, 2016-05-01:
The different sections in the Profiler can not be opened for me. 
Everything in "Button mappings" works fine and the profile works for my controller if I leave the profiler window open but none of the other sections can be viewed and so I am unable to map my axis to WASD.

Windows 10 x64.

RitzyBusiness, 2016-05-05:
Same here~ nothing in the profiler seems to be working.  Running the latest beta software~

AppleJewsy, 2016-05-11:
I can't get the profiler to work at all. Changing the mapping doesn't actually do anything, tested with MGSV and the input tester. Also swapping doesn't work inside of the profiler (switching the mapping for buttons x and y assigns y to act as x while x refers to y in a way that makes it still act as x, so I guess it doesn't refer to the actual button but rather its function)

closed 2018-01-26 by nefarius