nefarius/ScpToolkit #299: "Fake" DS3 detected; Genuine Sony DS3 detected; no Bluetooth

machstem, 2016-05-05:
First and foremost, this isn't my first rodeo and it seems like I'm doing everything I've always done to get this working.

I have 3 PS3 controllers, and all 3 were purchased from local places like Walmart, EBGames and Superstore.  

I can get all 3 controllers to work while plugged in by USB.  Only 1 of the 3 seems to work using Bluetooth.  The other two make the "sound" that they've connected over Bluetooth, but when looking at monitor (legacy through SCP Toolkit) I see their Bluetooth address as 00000000 yet the indicator messages say either "Fake DS3 detected" and the other says "Genuine DS3 detected"

The one that works, I only see SIXAXIS written on the device and its model number is:  CECHZC1U

The other two, the one that says "Fake" has both SIXAXIS and DUALSHOCK 3 written on it and its model number is:  CECHZC2U and in small lettering just beside it, A1

The one that says "Genuine" has both SIXAXIS and DUALSHOCK 3 written on it and its model number is:  CECHZC2U

I also noticed that on the "fake" one, when I unplug it from USB it doesn't "flash" and try to connect by Bluetooth, I have to press the PS button.  The "genuine" one works as expected; unplug from USB and it tries (but fails) to connect via bluetooth.

I'm not certain what I can do to provide more information in order to get more help, but again, I am certain all my PS3 controllers were purchased by legitimate stores and would expect the controllers to be legitimate as well.

Thanks for any advice and I will provide whatever is required to help troubleshoot this problem further.

machstem, 2016-05-05:
I just got them all prepped for a reinstall and I noticed something:

PS(R) Gamepad VID: 054C,PID: 0268

gunns, 2016-06-02:
Same problem here but mine appears as PLAYSTATION(3)Conteroller (VID: 054C, PID:0268)

bluetooth appears as PLAYSTATION(R)3Conteroller-ghic (in the log appears this way)

installed everything, when wired it works fine, but when i unplug it, it start flashing the 4 leds and in the SCP server notifications it shows "Genuine Sony DualShock 3 found"

here i attach a log of the scpServer, in timestamp 1464886095612 i pressed the PS button while unplugged, it "tries" to connect but something went wrong.

Thanks in advance.

[ScpDsxServer_gustavo.log.xml.zip](https://github.com/nefarius/ScpToolkit/files/295757/ScpDsxServer_gustavo.log.xml.zip)

dzfacts, 2017-01-19:
Is it possible to make it work with tha latest ScpToolkit  version (v1.6.238.16010), I got two of that PS(R) Gamepad , both of them work wired up, but they are not pairing. The genuine works as normal.

closed 2018-01-26 by nefarius