nefarius/ScpToolkit #304: Error causing SCPtoolkit to not recognize controller

gotmilk0112, 2016-05-13:
SCPToolkit has worked fine with my DS3 for the longest time now, but for some reason, it has stopped detecting the controller when I plug it in, and I have to run the "server" program in order for it to detect.

When I run the server program, the tray notifications properly pop up, but one of them is an error:

"Couldn't plug in virtual device 1: System.ComponentModel.Win32Exception (0x80004005): The parameter is incorrect"

The controller works, but it only works when the server program is running. As soon as I close it, the controller stops working.

What's going on?

johnmatricks, 2016-06-06:
Having this exact issue. Anyone know a fix?

gotmilk0112, 2016-06-06:
I just ran the "Driver Installer" and reinstalled the drivers. That seemed to fix it.

johnmatricks, 2016-06-06:
tried that as well as restarting and doing that again. Did not fix it.

johnmatricks, 2016-06-06:
actually, it did. Had to restart while it was still installed, thanks

closed 2018-01-26 by nefarius