nefarius/ScpToolkit #307: Cannot Install Driver on Windows 10

M3SS3NG3R, 2016-05-18:
Hi all,

I have been using ScpToolkit for a while on Windows 7.  No problems there.  However, recently I upgraded to Windows 10 (no hardware change whatsoever) and the driver simply refuses to install.  Specifically, when I run Scp Toolkit Driver Installer the window for user access control pops out asking for permission.  After that...nothing.  No new window, no error message.  I have tried to run as administrator and it didn't help.  It's driving me nuts!  Can someone help me out?

I have attached the log file for ScpDsxInstaller here.
[ScpDsxInstaller.log.zip](https://github.com/nefarius/ScpToolkit/files/271491/ScpDsxInstaller.log.zip)

Let me know if you guys need anything else!

damianshrinivas, 2016-05-19:
Hi. Did You check "force install" in SPC Toolkit Driver?

M3SS3NG3R, 2016-05-20:
Hi.  I don't think I can check that option because the driver installer won't run.  When I double click on it only the usual UAC dialog shows up and that's it.  Back in the days on windows 7 the driver installer would then proceed to ask you if you want to install bluetooth driver, DS3 driver etc etc that's where you can check "force driver installation".  On windows 10 the installer just quits silently (I checked task manager).

damianshrinivas, 2016-05-20:
Hmm it seems to me that You're missing some components. Check that You have installed 
.NET Framework 4.5 (it should be installed by Win10)
Visual C++ 2010 Redistributable Package
Visual C++ 2013 Runtime
DirectX Runtime

M3SS3NG3R, 2016-05-20:
Yeah I already have all of those installed and updated.  Can someone check if the log file has any clues?  Worst comes to worst I'll make a clean install of win 10 and see if that fixes it but just don't have the time or the energy to do that right now.  :(

M3SS3NG3R, 2016-05-20:
BTW, a search through the windows event viewer revealed this:

```
Faulting application name: ScpDriverInstaller.exe, version: 1.6.238.16010, time stamp: 0x569253fb
Faulting module name: d3d9.dll, version: 10.0.10586.0, time stamp: 0x5632d381
Exception code: 0xc0000005
Fault offset: 0x0000000000030ff4
Faulting process id: 0x1d54
Faulting application start time: 0x01d1b14ea8df5f0c
Faulting application path: D:\Program Files\Nefarius Software Solutions\ScpToolkit\ScpDriverInstaller.exe
Faulting module path: C:\WINDOWS\SYSTEM32\d3d9.dll
Report Id: 9ffd6680-b2ba-44e9-b3df-1e81709a6676
Faulting package full name: 
Faulting package-relative application ID: 
```

However DirectX is already fully updated (checked through both windows update and tried to manually update using DX end user runtime from microsoft).  Any ideas?

charlesmilette, 2016-05-20:
Try reinstalling DirectX 9 from https://www.microsoft.com/en-ca/download/details.aspx?id=8109

damianshrinivas, 2016-05-20:
Yeah, You don't have directx9 fully installed so install it from link above.

M3SS3NG3R, 2016-05-20:
Tried it and still no go.  Same error message in event viewer.

damianshrinivas, 2016-05-20:
https://www.microsoft.com/pl-pl/download/details.aspx?id=34429 and with this?

M3SS3NG3R, 2016-05-20:
That's for older versions of windows not for Win 10.

Serzhas, 2016-06-18:
I've hit the very same problem and have the same **d3d9.dll** error in Windows 10 Event Viewer. I have installed DX9 from link provided above, but error won't go. Strangely enough, putting **d3d9.dll** beside **ScpDriverInstaller.exe** did solve the issue...

closed 2018-01-26 by nefarius