nefarius/ScpToolkit #310: Controller not reacting, but pairing works

MisterFab1an, 2016-05-21:
I have a [3rd party controller](http://www.gioteck.com/controllers/sc-1/) and it connects when I install the drivers for DS4.
It will work for like 5 seconds and then freezes.
Debug is in the attachements!

[USB Controller_hid-report.dump.txt](https://github.com/nefarius/ScpToolkit/files/275632/USB.Controller_hid-report.dump.txt)
[ScpToolkit_Log-Package_MisterFab1an_1463792324.zip](https://github.com/nefarius/ScpToolkit/files/275633/ScpToolkit_Log-Package_MisterFab1an_1463792324.zip)

Jan200101, 2016-05-22:
> when I install the drivers for DS4.

Your controller  ( look at the [website](http://www.gioteck.com/controllers/sc-1/) you linked)
>  COMPATIBILITY
>Playstation 3

MisterFab1an, 2016-05-22:
I tried to install them for DS3 but they don't work with my controller. The toolkit will crash.
So I tried with DS4 and it works for so said 5 seconds but then freezes.

closed 2018-01-26 by nefarius