nefarius/ScpToolkit #313: Req: Expose Original PID/VID

fr500, 2016-05-22:
Is it possible to expose the original VID/PID to the OS?
For RetroArch we use those to match the controllers to the profiles, exposing the PID/VID would allow us to have accurate button names but it seems we always get the XBOX 360 PID/VID

RetroArch [INFO] :: Device #0 PID: {028E} VID:{045E}
RetroArch [INFO] :: Device #1 PID: {028E} VID:{045E}

fr500, 2016-05-22:
Or maybe just a baked in PID/VID different from the standard?

mirh, 2016-07-16:
\*Kind\* of related to #146

nefarius, 2016-07-16:
Do you mean the VID/PIDs of the original "feeding" hardware controller? Or different IDs for the emulated Xbox controller? Sorry, I don't seem to get it @_@

mirh, 2016-07-16:
He was talking of the original (probably so that he can show "PS" buttons) hardware, but I think this was kind of relevant to the same "abstracting" code of the other issue.

fr500, 2016-07-16:
Yes I mean the VID/PID of the original controller so apps that use VID/PID for profiles can have correct button names

nefarius, 2016-07-16:
Ah I see. But don't you face problems with a lot of devices just spoofing "genuine" VID/PIDs especially for gamepads?

nefarius, 2016-08-28:
I did dome testing and I can confirm that I in fact *can* provide the original IDs :smiley:

mirh, 2016-08-28:
Cool, I guess this kind-of-allow nice driver to still have games with [native](https://github.com/nefarius/ScpToolkit/issues/82#issuecomment-164045496) DS4 support work
(I use too often the word "kind")

fr500, 2017-01-08:
@nefarius pretty darn cool, I hadn't checked.
Are you doing so already?

closed 2018-01-26 by nefarius