nefarius/ScpToolkit #315: Analog stick randomly moving to the upper left

dankimxdd, 2016-05-24:
So every so often there will be an input to move towards to upper left. I don't know why its happening but I keep crashing jets because of it.
[ScpToolkit_Log-Package_Daniel_1464048160.zip](https://github.com/nefarius/ScpToolkit/files/278500/ScpToolkit_Log-Package_Daniel_1464048160.zip)

threydor, 2016-05-29:
Have you tried setting dead-zone treshold to higher value? It looks like a hardware problem to me (my analog stick is worn out and i had the same problem).

Diagonet, 2016-07-08:
So, made an account just for this issue. I had the same problem with my ps3 controller and no amount of dead-zone would make it better. I tried to use a brand new ps3 controller and the issue was gone.

Still, using the "faulty" controller with my ps3 does NOT cause any issues, which makes me think that maybe, while having a hardware side to this problem, it could also be fixed with ScpToolkit on the software side.

closed 2018-01-26 by nefarius