nefarius/ScpToolkit #316: An Unhandled exception occurred: System.NullReferenceException...

mannylouey2013, 2016-05-25:
Hi there. Firstly, I want to say thanks for writing and developing this new SCP toolkit. When it works (which is most the time) it is amazing. It feels great to just pick up my DS4, hit the PS button and instantly have it connect and be able to play. So thank you.

Here is my error message. I am unable to use my DS3 or DS4. I always  get an error like this.
The controller itself physically lights up, but has no response. SCP Monitor's "check input" option is greyed out, and the monitor itself shows the controller address as 0::0 lol.

I've attached both the debug tool output and the screenshot my error messages. Any help would be greatly appreciated. 

(Sweets is my computer's account name, DigitalBuddy is my PC name).
I'm running 64bit Windows 7. Let me know if there's anyhting else you need to know.
[image]
[ScpToolkit_Log-Package_Sweets_1464164540.zip](https://github.com/nefarius/ScpToolkit/files/281882/ScpToolkit_Log-Package_Sweets_1464164540.zip)

johnmatricks, 2016-06-06:
Ive had the issue on the right, not the issue on the left, would appreciate any info as well

johnmatricks, 2016-06-06:
nevermind, just restarted and it fixed

mannylouey2013, 2016-06-11:
To anyone else with this issue, I solved it by completely uninstalling and reinstalling SCP toolkit. I deleted everything that was left in the SCP toolkit installation folder too.

Not sure what happened, but it works now. Good luck to anyone else with a problem and remember, if your controller keeps blinking without connecting, you can use a pin to press a reset button on the bottom of it. it's there for both DS3 and DS4.

closed 2016-06-11 by mannylouey2013