nefarius/ScpToolkit #318: DS3 via USB Constant Vibration

wafflefaust, 2016-06-01:
Hello, I've just spent the better part of a few hours trying to get the old Scp-DS-Driver-Package-1.2.2.175 running on my Windows 7 machine. I've tried my best to clean up the old installs and follow the old uninstall directions before installing the most current version, (checked via ScpToolkit Updater), but something has gone wrong and started after I pushed the replace driver button on the 2in1 USB Joystick driver in Zadig. Ever since then the controller will not stop vibrating, and sends no key data, (checked via Monitor program.) 

Do I need to run the cleaning program? I've stopped fumbling with my inadequate troubleshooting attempts, thinking I'm doing more harm than good at this point. 

I'm using a wired ps3 controller, so no bluetooth. I also don't have a log file because there is an error when I run the debug info collector: 

"ERROR: The process cannot access the file 'C:\Program Files\Nefarius Software Solutions\ScpToolkit\Logs\ScpDsxMonitor_Wafflefaust.log.xml' because it is being used by another process."

wafflefaust, 2016-06-01:
Here are the logs I do have, was having a lot of difficulty trying to drag and drop the archive to this folder so I put them on my dropbox: https://www.dropbox.com/s/w3rxz5v058lwnji/Wafflefaust_logs.ZIP?dl=0

nefarius, 2016-06-11:
[Have you tried the manual removal?](../wiki/Manual-driver-removal)

wafflefaust, 2016-06-13:
I just followed the steps for manual removal, and uninstalled, rebooted and then reinstalled SCPToolkit. Now I feel like the installation is broken because the monitor app doesn't even load, and the settings app gives this error: 
"Couldn't fetch configuration
Couldn't request configuration, make sure the SCP DSx Service is running!"

So I went in and checked, the service is not running. Next I tried right clicking on the "SCP DSx Service" and selecting "Start" but get this error:

"Services
Windows could not start the SCP DSx Service service on Local Computer.

Error 1068: The dependency service or group failed to start."

wafflefaust, 2016-06-13:
Maybe the Virtual Bus Driver didn't get re-installed?

wafflefaust, 2016-06-13:
Do I need to remove the Xbox 360 Accessories program before installing this?

wafflefaust, 2016-06-14:
I've repeated the manual removal and install process a few times now and feel that there's something going wrong when the driver install app tries to attach the libusb drivers to the device. I've made a new set of logs for your reference: 

https://www.dropbox.com/s/sdob91iz1zxtj3j/ScpToolkit_Log-Package_Wafflefaust_1465936217.zip?dl=0

wafflefaust, 2016-06-28:
Hello Nefarius, I've run the clean wipe utility and am in the middle of running the latest 1.7 beta. I did need one of the prerequisites, and got it installed, but have encountered a difference from your video tutorial. The time stamp is 12:21, and is at the point where I run the driver install utility. There is no device in the "uninitialized dualshock controllers" box. 

Like your comment on the video said, I believe this is because of the 3rd party (unofficial) ps3 ds3 controller. It's made by Arsenal Gaming if that helps any. Given the amount of time it's taken to try and get this thing working, I should have just bought an official controller and not had the headache this is giving me.

nefarius, 2016-06-28:
Thanks for the dump file, that really solved the mystery; that controller is in no way DualShock compatible so no wonder it goes crazy. I could add support later in the future.

wafflefaust, 2016-06-29:
Thank you for keeping up with this and for making this software. I can tell you've put quite a bit of effort and time into it. I'm sure it's quite frustrating when problems like this arise, the controller is definitely  PS3 compatible, but they must send button data differently? I wish I could just simply remap the buttons via a control panel. The constant vibration issue has gone away since installing 1.7, I think you mentioned something about using a signed Windows driver instead of the libwdi offering, so maybe that's what fixed it? 

I wish I knew more about the technical difficulties and capabilities when getting these drivers to address the controllers properly. It would be cool to have an option during the install process to capture controller data like the game pad analyzer does, and create a driver that fits.

h3mlocke, 2017-02-01:
I have a Snakebyte Wired PS3 controller sb00566
I am experiencing the same problem, is it incompatible as well?

closed 2018-01-26 by nefarius