nefarius/ScpToolkit #320: Xbox emulation support bug

ELAD3356P, 2016-06-01:


charlesmilette, 2016-06-01:
Have you tried with a Xbox 360 controller? Because Scp emulates Xbox 360, not Xbox One.

ELAD3356P, 2016-06-01:
Xbox one and 360 works almost the same on xinput in windows 10

But it's not the problem, i said that the ds4 controller is doing problems

The xone works great in those games.

But those games don't  recognize the 'fake' xbox 360 controller . 

You with so far?

giorgosp, 2017-02-15:
Any news about this? Controller is not recognized by Fifa for example but it is recognized by Far Cry 4

giorgosp, 2017-02-15:
I guess Fifa and similar games don't "want" xinput but directinput? I don't know about drivers, however I installed x360ce as well and now Fifa is fine :)

closed 2017-10-02 by ELAD3356P