nefarius/ScpToolkit #323: Make it act like an mouse ?

d3xt3r01, 2016-06-03:
1st things first, Awesome piece of work !!!!
Now to the funny business:

Is there a way to make it act like a mouse ? Not in the traditional sense ! I'd imagine trying to use it on slither.io ( online game ).
I don't think it would require much... Having a circle with the center of the joysting being the center of the screen and the mouse would be in whichever position the joystick is in ( given the radius of the circle can be defined .. no need for it to go crazy all over the place ).

While at this question .. even if using it completely as a mouse .. I'd imagine mapping buttons to various keys would be doable ..

Thanks in advance.

charlesmilette, 2016-06-03:
[JoyToKey](http://joytokey.net/en/) can bind controller buttons to keyboard buttons or mouse movements.

d3xt3r01, 2016-06-05:
Yeah, I guess that would work .. Thanks.

closed 2016-06-05 by d3xt3r01