nefarius/ScpToolkit #324: BSOD caused by ScpVBus.sys

Ratebite, 2016-06-03:
I had ScpToolkit working for a while until I suddenly started getting BSODs caused by ScpVBus.sys whenever the DS3 service is running and a DS3 controller is connected (I will crash as soon as Windows and the service starts if I leave a DS3 connected).  I had a lot of problems after trying to reinstall the toolkit (I couldn't even run the debug info collector), but I think I eventually returned to the original problem.  I ran the debugger as instructed (I hope) and have uploaded the logs.  Any help would be greatly appreciated.  

https://www.dropbox.com/s/ffj4ge73gqosyr1/ScpToolkit_Log-Package_Yeshaih_1464992849%20%28drivers%20installed%2C%20service%20stopped%2C%20DS3%20connected%29.zip?dl=0

mirh, 2016-06-21:
This used v1.6.238.16010. 
Can't you test newer one?

Ratebite, 2016-06-24:
I've tried v1.7.276.16101-BETA, but I'm unable to complete the setup (which might be because I don't know if I was able to completely wipe everything out).  I was able to install the toolkit, I was able to install the prerequisite Microsoft files that I was missing for this version, I have still have Microsoft's Xbox controller drivers installed, and I was able to initialize my DS3 controller.  I'm not using Bluetooth so I skipped that part in the driver installer.   I start having problems with the Xbox 360 Controller emulator driver page.  Along with an "installation failed" window that opened with a few suggestions, these are the pop-up messages I received when running it that appear in the bottom right corner: http://puu.sh/pEEvt/2ad8db77a7.jpg.  
I tried to continue anyway (which is probably a mistake) and was able to install the Windows Service in the next part of the driver installer.  I tried running the installer and restarting my computer several times, but I can't seem to even install ScpVBus with this version.  I ran the debug info collector and uploaded the logs even though I'm not sure how useful they will be: https://www.dropbox.com/s/06jhtmx5klhz1zo/ScpToolkit_Log-Package_Yeshaih_1466796406.zip?dl=0
I appreciate your response.

mirh, 2016-06-24:
And ScpCleanWipe after normal uninstall doesn't help?
Also, I hope you weren't running [mij](http://pcgamingwiki.com/wiki/Controller:DualShock_3#Removing_MotionInJoy) before.

Ratebite, 2016-07-11:
Using the ScpCleanWipe never helped unfortunately, I've tried uninstalling and reinstalling more times than I could count, but I still get the same BSOD.  I've never used MotionInJoy on this hard drive.

nefarius, 2016-07-16:
[Have you tried this already?](../wiki/Manual-driver-removal)

closed 2018-01-26 by nefarius