nefarius/ScpToolkit #333: Cant conect Bluetooth for third party PS3 dual

wssfisio, 2016-06-14:
Hi, as i checked i have a bluetooth compatibe dongle and third party compatible controler but cant conect with bluetooth. At monitor i just get the anoying 0000000 Bluetooth None. Can you help me please? Thanks in advance.

[ScpToolkit_Log-Package_wssfi_1465876918.zip](https://github.com/nefarius/ScpToolkit/files/313108/ScpToolkit_Log-Package_wssfi_1465876918.zip)

XantheFIN, 2016-06-19:
Hello. If it is in case of multiple controllers. Try reset controller by pressing under the controller with a pin that reset button.

I have same problem with only one controller and trying find solution too.

arekhn, 2016-06-22:
I have the same problem, but i dont understand your explanation, download scarlet crush version, with update and then moded scpcontroll

then with an alredy scpserver from nefarius version, start old version?? i just dont get it

XantheFIN, 2016-06-24:
Hi, sorry being unclear.

So for now i think i need after pairing correctly and until i hit with (nefarius) scpserver to that 0000000 Bluetooth None -error.

I then stop by hitting "stop" and start this orginal [1.2.0.160] scpserver with modded ScpControl.dll being there and hit Connect and then try connect PS3 controller and it does work for me.

arekhn, 2016-06-24:
OMG you are right, it is working!!! :D Thanks man, this is great, have you tried only using the scarlet crush version? (with no Nefarius stuff)

XantheFIN, 2016-06-24:
No and i have had idea trying fix Nefarius version so it works without problems but so far i have had personal problems and hard try to focus on this or anything.

Sorry.

Maybe soon i can focus on this enough. He had made big chances example while loop has been removed and replaced elsewhere which i havent yet looked..

closed 2018-01-26 by nefarius