nefarius/ScpToolkit #335: Support up to 15 Controllers

bigbear2nd, 2016-06-14:
Hi!

Thank you for putting all this work and thought into the DS3 controller driver!

I would like to request to increase the maximum of the supported controllers. I really like the controller and there are many 6 players and more games out there.

I could imagine the leds of the controller to show the controller number digitally 0001, 0010, 0011, and so on. :)

Thanks again and cheers
Bob

mirh, 2016-06-15:
It might be even possible.. But remember XInput is limited to just [4](https://msdn.microsoft.com/en-us/library/ee417001.aspx#multiple_controllers) players. 
DirectInput has [no](http://microsoft.public.win32.programmer.directx.input.narkive.com/6D6uyWxl/maximum-number-of-directx-directinput-devices) such limitation, but you are still limited by whatever protocol logically carry your data. And iirc Bluetooth is limited to [7](https://en.wikipedia.org/wiki/Bluetooth#Communication_and_connection) devices. 
For as much as USB [supports](https://communities.intel.com/thread/52417#239236) tons more, a seamless experience probably require a lot of fine tinkering.

bigbear2nd, 2016-06-15:
Thank you for your detailed answer. I might happily purchase another 2 or 3 Bluetooth dongles, if Scp could handle them :smile: 

I guess many other players might also do and like this.

I guess it that Scp takes the XInput for easy management? Is the DirectInput you mentioned difficult to handle? If so, i would try to get the support upstream and file an issue there. :smile:  Any directions where i could do this? 

I tested Scp with 4 PS3 controllers and 2 USB (Cable) controllers with no problems. :smile: But it would be more fun/fair to have all the same controllers. 

Chinese manufacturer of ps3 controller copies for pc claim to support up to 20 controllers on one pc. 

Best regards
Bob

nefarius, 2016-07-16:
It heavily depends on the games you use. The main purpose of the toolkit is to make the DualShocks available through XInput while the DirectInput support is a nice side-effect of Microsoft`s own XUSB driver. I'm currently experimenting with lifting the limits of supported controllers but then again; the current API limit in XInput (which every game with Xbox controller support uses) if 4 controllers. That's not in any way a hardware limitation but a limit that Microsoft themselves forced onto developers if they only use XInput.

bigbear2nd, 2016-07-20:
Hi!

Thank you for tinkering with the increasing of the supported controllers. 
Is it possible to have 4 controllers with XInput and 11 with DirectInput? ;)

I can supply you a list of games which profit from more than 4 controllers. Would it help?

Best regards 
Bob

nefarius, 2016-07-20:
@bigbear2nd in my current development I could theoretically support tens or even hundreds of emulated controllers but still; games using XInput won't profit from that.

Wouldn't hurt if we had a list on games :)

closed 2018-01-26 by nefarius