nefarius/ScpToolkit #342: BT Drivers issues in Win10

hige-agusd, 2016-06-23:
I can connect via USB with no problem.
When I install BT drivers for my Intel(R) Wireless Bluetooth(R) (VID:8087, PID:0A2A), nothing happens.
One I reboot, my controller connects via BT, but I can't use the BT device to connect to anything else (and won't appear under my Devices and printers, as if I didn't have Bluetooth)

Do you need any more info?

charlesmilette, 2016-06-23:
That behavior is normal. That is why it is recommended to have a dedicated Bluetooth dongle for the controllers.

zelmon64, 2016-06-23:
@hige-agusd Which controller are you using? If it's a DS4 I was happily surprised to find that it works natively (more or less) meaning that you don't have to slave a BT adapter to it.

hige-agusd, 2016-06-23:
Unfortunately, I have a DS3
It seems I'll have to get a dongle
Thanks for the replies, guys!

pacoesni, 2016-06-24:
I'm maving the same problem, even if I use two Bluetooth devices. My bluetooth-in devices take the driver from scp and use it, even if I select just install drivers for the external dongle.

closed 2018-01-26 by nefarius