nefarius/ScpToolkit #345: XBOX 360 controller driver failed to install

Madeiner, 2016-06-25:
Hi there.
I used this program to great effect a few months ago on my PC.
Now i'm trying using it on a different pc, but it doesn't work as expected.

As soon as i install the "XBOX 360 controller driver" from this software, windows tries to reach for windows update and update the driver. It fails (even if i cancel) and i'm left with a PS3 controller in device manager, and a nonfunctioning XBOX 360 controller with a yellow mark. Nothing shows up in usb game controllers window.

I've tried installing and reinstalling, even with forced installation. Also tried SCP wipe utility.
Any help?

closed 2016-07-17 by nefarius

Dienfive, 2017-07-22:
Hello Madeiner,

Do you know how to fix this issue? as i have same issue but dont know how to fix it:(

Madeiner, 2017-07-24:
Hello,

it's been a while but i can try to help you.

If you have any software related to the joystick, try to uninstall it and
run a driver cleaner.
Then download this one:

http://forums.pcsx2.net/Thread-ScpToolkit-XInput-Wrapper-aka-ScpServer-Reloaded

It's different from the one i was using at the time (even if very similar),
and worked flawlessy multiple times on different computers.