nefarius/ScpToolkit #349: SCP needs to be reinstalled after each computer restart

Iking47, 2016-06-28:
SCP works fine but it doesn't work each time after I boot up my PC (Windows 10)  and thus needs to be reinstalled. It doesn't take long to reinstall it but isn't it supposed to run from Windows boot?

The SCP DSx Service is still checked in msconfig under services and SCP Toolkit Tray Notifications are enabled in the startup list.

keithomalley, 2016-07-02:
I get this error as well, using a bluetooth dongle and a PS3 controller, I tried checking "Force Install" and the installation just hangs, but if I install normally every time I boot then it works again. It doesn't always happen but has been the last few times.

Possibly of note - I've been leaving my controller plugged in by USB cable while I boot, though I haven't tested this

charlesmilette, 2016-07-02:
If you're using Bluetooth, it may be related to #51

mirh, 2016-07-04:
Did you try to put SCP service to "Late" start as suggested [here](https://github.com/nefarius/ScpToolkit/issues/224#issuecomment-192291387)?

keithomalley, 2016-07-05:
@mirh That's fixed it for me! thank you :)

mirh, 2016-07-05:
Ugh. 
I'd try to bump all threads, let's see.

ikwtif, 2016-09-24:
I seem to be having the same issue. I have to reinstall everything on a restart. Setting the scp service on delayed start hasn't helped

Artuvazro, 2016-09-27:
I have this problem too. When the service is stopped and then restarted, the controller won't pair via Bluetooth unless I reinstall the whole thing.

charlesmilette, 2016-09-27:
@Artuvazro does it works via cable?

Artuvazro, 2016-09-27:
@charlesmilette Yes, with the cable it works without any need to reinstall the driver. But when I disconnect the cable it doesn't pair via Bluetooth unless I reinstall the drivers again.

charlesmilette, 2016-09-27:
Then #51 is your issue

Artuvazro, 2016-09-27:
@charlesmilette  Oh, thank you ;)

closed 2018-01-26 by nefarius