nefarius/ScpToolkit #350: SCP broke my bluetooth drivers

subzerocro, 2016-06-28:
I had a bluetooth mous and after installing SCP drivers for my dualshock it does not want to work. Niether does my joystick want to work now and it did before. (i tried to switch the port it conncted to.

Now i can use my mouse.

subzerocro, 2016-06-28:
NONE of my ports want to use my bluetooth mouse and the 2 front ports i have dont want to take my wired mouse.

zelmon64, 2016-06-28:
ScpToolkit slaves the bluetooth adapter for use with the PS controllers. You'll have to reinstall the normal bluetooth driver to get your bt mouse to work. 

For your wired mouse, sometimes it take a while for the drivers to install on a new port so just leave it in for a bit and wait .

subzerocro, 2016-06-28:
Thank you! But i dont know how nor where to reinstall the bluetooth drivers

zelmon64, 2016-06-28:
There's probably a way to do it with the software but I like the device manager. On win 8.1 or 10 right-click the windows logo and select Device Manager. Then find the bluetooth adapter and uninstall and delete the driver. You can check that you found the correct device by unplugging it and watching it disappear from the manager. Scan for new hardware in the Device Manger or unplug and re-plug the adapter to make it install the default driver.

If you get two adapters there's a way to slave one to ScpToolkit and use the other normally but it's a little fiddly.

subzerocro, 2016-06-28:
I have windows 7 ultimate, and there are no bluetooth drivers in device manager. I already tried to do what you said.

zelmon64, 2016-06-28:
No it won't be under Bluetooth - that's the problem. For the original version it was under "Universal Serial Bus devices" so it's probably something similar for this version of ScpToolkit. Just unplug and replug your adapter a few times to see what changes in the Device Manager.

subzerocro, 2016-06-28:
Fixed my mouse. But scp doesent work anymore and I cant play rocket league with a controller anymore

subzerocro, 2016-06-28:
Fixed my mouse. But scp doesent work anymore and I cant play rocket league with a controller anymoreads

subzerocro, 2016-06-28:
My controller doesent work anymore tho. Fixed my mouse

subzerocro, 2016-06-28:
scp fucking sucks

Jan200101, 2016-07-12:
really ? really.... 
cant you read ? @subzerocro 

> ScpToolkit slaves the bluetooth adapter

subzerocro, 2016-07-12:
Well no shit!! I knew that, the porblem was that it didnt want to fucking reset. And great job posting to a 2 week old topic!

nefarius, 2016-07-16:
LOL **two weeks** old...

[image]

merklaw, 2016-07-21:
There is insufficient information regarding the problem in this ticket. You are welcome to close the ticket, nefarius.

closed 2016-07-21 by nefarius