nefarius/ScpToolkit #351: SCP - not working

subzerocro, 2016-06-28:
I installed everything correctly (using the older SCP - zipped SCPserver). Everything installs correctly exept when its done the c ontrollers blink like they are charging and than stop and dont want to work In game. Is there any fix for this?

wafflefaust, 2016-07-01:
Exactly what version of ScpToolkit are you using? Older could mean the abandoned Scarlet.Crush project, or older revisions of Nefarius' work. Have you tried the latest 1.7 beta? Nefarius has also made a tutorial video walking us through the steps. I highly recommend the 1.7 beta as it does a prerequisite check on your machine to ensure you have all the additional official Microsoft libraries and such installed.

Here's a link to the tutorial video:
[https://www.youtube.com/watch?v=yH0pTK62DAQ](url)

Here's a link to the 1.7 beta (top most link is the latest):
[https://github.com/nefarius/ScpToolkit/releases](url)

subzerocro, 2016-07-12:
Nothing works anymore. I got a PC joystick so i dont have to use SCP i cant use it. Great!

subzerocro, 2016-07-12:
SCP DS3 Service is not installing and I cant use my countroller without it. Everything used to work but now it does not want to

closed 2018-01-26 by nefarius