nefarius/ScpToolkit #352: Fatal - Couldn't open the device (DS3 after plugged in via USB)

LoneEch, 2016-06-28:
I've been running into a problem with SCP toolkit, whereas every time I plugged in my PS3 controller after installing its drivers (and running the driver installer, then restarting) it would give me an error stating that it "Couldn't open the device".  I've used the clean wipe tool, done the full procedure to uninstall SCP's drivers and services as a reassurance, and I've gone through this several times.  I had used Motioninjoy in the past, but I have followed the full removal instructions given by the [Emulation General Wiki](http://emulation-general.wikia.com/wiki/MotioninJoy) in order to clean it out of my computer fully (as far as I know at least).  I've also uninstalled the Xbox controller driver and reinstalled it multiple times, too.

Even with all of these steps, after installing the DS3 drivers from SCP, it still gives me the same error, and prevents me from using my PS3 controller.  I do not use Bluetooth, but it also states that it cannot reach the Bluetooth host address for my controller.  It's been plaguing me for a long while now and I really just want to play Rocket League. ;_;

Here's a screencap of the SCP Server log:
[image]

along with a screencap of SCP monitor showcasing the port:
[image]

I'm not too sure what else I should add into here, but if I need to please do tell me.

Darkholme, 2016-06-29:
I've been having the same issue, but I wasn't using Motionjoy before. However, I have been trying to get my controllers to work all evening (two official dualshock 3s). I tried installing this suite, and wasn't getting any errors, but nothing would detect any input from the controllers. I uninstalled and then tried the old version of SCP. It didn't work either. I then reinstalled this version, and started getting these errors.

LoneEch, 2016-06-30:
Problem is, I've tested to see if that might have been the case by installing with and without the Bluetooth driver setting ticked for installation.  Both situations produced the same result, that being the errors I've been getting when I do insert the USB into the port with the PS3 controller connected.  I don't know how many permutations of these conditions I've tested so far, but none have come up with anything positive. :^(

Here are the errors notifications in subject, before I forget to add them here again:
[image]

Darkholme, 2016-06-30:
Yep those are exactly the same messages I was getting whenever my controllers were plugged in via USB.

Here's what *I* did to resolve it.

1. Run the SCPCleanWipe Utility.
This should make the messages stop popping up when you plug them in (I would test to be sure here).
2. Reinstall my Bluetooth adapter (I *WANT* My controllers to work via bluetooth). This may not matter to you.
3. Run the SCPDriverInstall Utility.
3.1 Uncheck Dualshock 4, unless you're going to be using Dualshock 4 Controllers.
3.2 Choose Bluetooth Dongles
*ONLY* Select your bluetooth adapter, and only if you want your controllers via bluetooth instead of USB. Otherwise, Uncheck the Bluetooth Driver button.
3.3 Choose Dualshock 3 Controllers
Select *ONLY* any PS3 controllers in the list.
4. Click Install, and Wait.
5. Enjoy your controllers.

LoneEch, 2016-06-30:
Your step-by-step procedure is appreciated, but sadly it doesn't seem like we have the same case.  The same notifications pop up after all of those steps, and the problem persists to show in the server logs.  :^(  Every time I install the drivers, though, I do get a NullException notification pop up, so I'll see if I can get that screencapped.

LoneEch, 2016-08-24:
Turns out that all I needed to do was reset the PS3 controller while it was plugged in.  Strange, but I guess this is solved somehow.

closed 2016-08-24 by LoneEch