nefarius/ScpToolkit #353: Bluetooth connection to controllers isn´t working after reboot.

EnderCraftDE, 2016-06-29:
I recently updated the version v1.7.277.16103-BETA, because the previous versions had the constant error, that the service "Ds3Service" couldn't launch after a reboot, so the entire program wasn't responding correctly. 

Now the program seems to hate me:

1. 
Usually, the service starts properly, but about every third time, it just doesn´t work, like shown here:
[image]

2. 
The Bluetooth connection to my controllers doesn´t work all the time, even if it seems to be launched properly and the service is running flawlessly:
[image]
(My main controller is connected via USB, but my second one couldn´t connect at all.)

The Bluetooth driver should be installed correctly too, I think:
(I have an integrated Bluetooth card. The Hardware-ID is "USB\VID_04CA&PID_300B".)

[image]

The Virtual Bus Driver is also installed correctly.

I hope, someone could help me because I don´t know what to do next, in order to get this working.

Best regards,
EnderCraftDE

Btw, I hope my English isn´t too bad.

charlesmilette, 2016-06-29:
See #51

mirh, 2016-07-05:
Not that I'd like to trigger any anti-spam filter.. 
But try to put SCP service on "Late" start.

Jan200101, 2016-07-12:
Set SCP service to late start as mentioned above and set it to restart if it crashes.

I can give you a tool if you cant do it yourself

EnderCraftDE, 2016-07-24:
Putting the service on Late start, just fixed the problwm twice, later it was like before.

closed 2018-01-26 by nefarius