nefarius/ScpToolkit #354: Installed all drivers, keyboard and mouse no longer do anything

lordjeebs, 2016-06-29:
I had installed and used scptoolkit before, but I had to reinstall Windows due to an unrelated problem. However, after installing scptoolkit again, it wouldn't work. I stupidly decided to install every possible driver, and now my keyboard and mouse (both usb) will not work. I really want to fix it, but I don't know how. If you need any more info, please let me know.

charlesmilette, 2016-06-30:
Use the clean wipe tool or [do a manual removal](https://github.com/nefarius/ScpToolkit/wiki/Manual-driver-removal).

Then restart the computer and reinstall ScpToolkit without installing every possible driver.

wafflefaust, 2016-07-01:
When you say you "stupidly decided to install every possible driver," what do you mean? Did you select each usb device that wasn't your controller and force the driver install on them? Do you have any ability to take screen caps of what you did that was "stupid?" (eg. repeat the process, but don't actually install the driver.)

Also try plugging the kb and mouse to a different usb port, sometimes that helps install the right drivers when it's recognized as being in a new usb port.

closed 2018-01-26 by nefarius