nefarius/ScpToolkit #358: Scp heavy input delay with wired rock candy ps3 controller

Dablackjohn, 2016-07-03:
I have 2 wired controllers, a Madcatz ps3 wired controller (works just can't use it well because of a really bad short). I recently bought a rock candy ps3 controller and its recognized by ds3 and installs drivers but there's a extreme delay in the inputs and i can't figure out why its doing that, basically I can't play with it. Also is there any tutorial for getting the official ps3 wireless controller to work with ds3? I used to use it a while back but its not fully attaching. Whenever i hook my ps3 controller (with usb cord) up to my pc it won't stop blinking (signalling its not connected or something)

merklaw, 2016-07-21:
So you have two controllers. You shouldn't expect the controller with the broken cable to work if it has a short. The other controller is a third party controller, and it would really help if you could be more specific about how you perceive the delay, like how long is the delay? Have you tried multiple applications/games and does it happen in all of them? Could you tell us the product ID of the controller? How many lights are blinking on the controller? One or four lights? Please provide as much information as possible when making a bug report.

Dablackjohn, 2016-07-21:
I'm well aware the one with the short won't work too well, it still works
but the cord has to be angeled a certain way.

By input delay only 1 button on the face buttons are working, its going
really slow. Its not reading any button inputs until 5-10 seconds later.
It's not being read as a 360 controller and ds3 isn't recognizing the
controller. No lights are blinking for this rock candy ps3 controller for
some reason. Out of the 4 buttons [] /_\ X and O. Only x works on games
that can be used with a 360 controller. The D-pad, triggers and shoulder
buttons etc.. Do not work. Nothing is working on it for some reason.

closed 2018-01-26 by nefarius