nefarius/ScpToolkit #362: Tray app interrupts game while updating

Alkaris, 2016-07-04:
The Scp Toolkit updater tray app is very annoying when it interrupts games while updating. Could this perhaps be changed so that it updates silently in the background so it doesn't do this? It's very annoying when playing a game and then all a sudden game stops just to show me the updater has completed.

If tray app is active, do not make it pop-up a window dialogue that interrupts everything else. An OSD notification will suffice enough to let person know it updated.

CREEATION, 2016-07-06:
Is there any option to disable auto-updating? I don't even want auto-updates and yes, it's more than annoying.

Alkaris, 2016-07-06:
@CREEATION - I don't think there even is option for that, except for perhaps removing it from Start-Up Applications folder. But I don't think that would even stop it.

RobinQuetin, 2016-07-13:
The SCP Toolkit Updater is started by a scheduled task named 'updater' ([where to find?](https://technet.microsoft.com/en-us/library/cc721931(v=ws.11).aspx)). You can customize the update frequency and other things (I think there is even an option which can hide the task, but I don't know if it still updates the application using this option), or even disable it completely.

Alkaris, 2016-07-13:
@RobinQuetin - just changing the scheduled task isn't enough, it should be made to silently update in the background and given the user a OSD Notification instead of dialogue box popping up on the screen.

nefarius, 2016-07-16:
The updater behavior is tamed down since 1.7

closed 2016-07-16 by nefarius