nefarius/ScpToolkit #366: Classic ABXY layout

Jop-V, 2016-07-04:
This is how the buttons are mapped:

```
○  →  B
×  →  A
□  →  X
▲  →  Y
```

Makes sense for a default because it matches the Xbox ABXY locations.

[image]

But I'd like to use the classic ABXY layout instead of the Xbox layout. Like this:

[image]

```
○  →  A
×  →  B
□  →  Y
▲  →  X
```

Could you add this as an option to the settings manager?

mirh, 2016-07-04:
If you want to switch button mapping, use x360ce.

Jan200101, 2016-07-12:
SCPToolkit is just translating your DS3 controller to a XBOX controller

nefarius, 2016-07-16:
@Jop-V this would be achieved by the new profile manager which is stuck in development hell since months =3 So sadly currently not possible without 3rd party tools.

mirh, 2016-07-16:
I'd see the sense if SCPtoolkit wanted to really go universal (#243 and 85cc28dd30ff9489da047b82f4b8cd7c743c307c) but tbh, until it isn't that far I think efforts would be better spent elsewhere.

nefarius, 2016-07-16:
@mirh [I guess you will love this](../../ViGEm) once it's finished ;)

mirh, 2016-07-16:
I'm dreaming of your tool to be like: 

																				↗ DirectInput
    HID → Raw Input API (or whatever standard needed to make #82 great again)
																				↘ XInput

nefarius, 2016-07-16:
With more and more knowledge of the wonders of the [KMDF](https://msdn.microsoft.com/en-us/library/windows/hardware/ff557565(v=vs.85).aspx) I also grow more and more powerful!

[image]

mirh, 2016-07-16:
Thinking to the aforementioned evil scheme, #290 last comment and #243.. I wonder if the whole thing couldn't become.. like modular?
There's a toy that converts standard Raw Input to DirectInput, one to XInput, then every supported controller would have one of those USB<->HID driver you mention.. And then I guess a pretty bare "profile" could be made that maps HID signals to RI ?

nefarius, 2016-07-16:
What I've learned from tinkering with device driver development is, that whenever you can you should incorporate with the existing Microsoft drivers so errors and development overhead is minimized. I like the modular approach; [in my rewrite](../../ViGEm) of the `ScpVBus` I already designed it with driver-to-driver communication in mind. This would allow to easily (with a few lines of code) add XInput support to [HID-Wiimote](../../../jloehr/HID-Wiimote) and other projects.

nefarius, 2016-08-10:
@mirh [Your wish is beeing implemented](https://www.youtube.com/watch?v=lALPkzTA-5Q) :smiley:

mirh, 2016-08-13:
\**Unzips*\*

nefarius, 2016-08-28:
[image]

mirh, 2016-08-28:
[image]

charlesmilette, 2016-08-28:
[image]

closed 2018-01-26 by nefarius