nefarius/ScpToolkit #367: Afterglow AP.2 wireless PS3, anyone have luck getting this controller to work?

rydholm, 2016-07-05:
The drivers installed just fine, and when I plug in the USB wireless dongle it recognizes it as an afterglow controller and gives a message like workaround applied and all positive messages, no errors.  But I get no input response from the input tester or anything else I try. Do wireless controllers of this type(USB, non-bluetooth) even work with SCP? It's possible my controller may be broken but I don't see anything to indicate it.(It keeps a charge, lights come on and everything) If I can't figure it out I'll have to find a friend with a ps3 and see if the controller works on that. 

Thanks in advance to anyone who answers.

SariaFace, 2016-07-10:
rydholm, on Win 10? I have a similar problem with Nyko Raven, which is also a USB wireless dongle type. It worked a treat on Windows 7, but after upgrade to Window 10 I get a similar issue where it seems to be read properly but no input. Did full uninstall/reinstall of newest SCP to no avail.

rydholm, 2016-07-11:
Yeah this is on Windows 10. my regular DS3 works just fine. I just can't get this wireless one to work.

closed 2018-01-26 by nefarius