nefarius/ScpToolkit #368: Playstation 3 controller does not work even it is recognized

evildog1, 2016-07-06:
Any games recognize Playstation 3 controller as Xbox controller, as it should normally, but pressing any buttons on Playstation 3 controller does nothing. While i install the driver, i got a bunch of notifications and i saw some errors like this:

[image]

[image]

No driver errors, no hardware errors. I have disabled the antivirus before install the driver, but after i re-enable it again, Antivirus detect SCPSERVICE.EXE and SCPMONITOR.EXE as virus and it even found new virus from C:\USERS\xxxxx\APPDATA\LOCAL\TEMP\PSSCC5A.PS1 that i have never seen before.

http://www.mediafire.com/download/a3118g1k6o4fw6w/ScpToolkit.zip

ItalisLoL, 2016-07-06:
Hey, I just starting having this problem.  I was playing some games on steam when all of a sudden my comtroller completely stopped working altogether. The Monitor tool shows that it is recognizing my inputs, but no games or anything else will show that it's really working. http://www.mediafire.com/download/am0z8wma74vozf6/ScpToolkit_Log-Package_yes_1467841927.zip

evildog1, 2016-07-07:
@ItalisLoL can you try older version of SCPserver?
That version is stable and it can be used for longer time like a year, but it did not work for me at this time. i don't know why :(
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller

Dablackjohn, 2016-07-18:
if its the official ps3 dualshock 3 controller, press the reset button on the back of the controller and it should work.

evildog1, 2016-09-17:
Well the latest version of ScpToolkit is working perfectly

Thecloudbacon, 2017-03-04:
I cannot use it even though it recognizes it. I get the error: --Bus Open: Offset 0 failed

xanscorp, 2017-08-21:
Nope...definitely not working perfectly.  After a period of inactivity, the only way to make the software work correctly again is to fully uninstall it (and any remnants in the system) and reinstall. So, yeah, on what planet is that "works perfectly" because it's certainly not this one.

xanscorp, 2017-08-27:
Sounds like some asshats with enough GitHub accounts screwed it up then approved the pull to screw everyone.  This shit has worked maybe once for me.  Since then SCP drivers have been worthless in getting my dual shock controller to work.

evildog1, 2017-08-28:
Excuse my english. That's how I say when it works for a year and i'm not having issue anymore. I only use PS3 controller connected with cable.

evildog1, 2017-08-28:
close

closed 2017-08-28 by evildog1