nefarius/ScpToolkit #370: Controller number is 2?

lamelizard, 2016-07-08:
Thanks for this great tool.

I'm using 1.6.283.16010, Windows 10 64bit and a genuine DS3 in wired mode
I got the issue that the virtual controller get's connected as controller number 2;
The Xinput Controller Tester, which came bundled with the old ScpDriver, has it's own XInput1_3.dll. With this dll in the same directory it states that controller number 1 is connected.
Without the dll, therefore using the one provided by Windows, it says that controller 2 is connected and controller 1 is disconnected.
Why is Windows connecting the virtual controller as Xinput device 2 if I have no over controller (The "Game Controllers"-Window shows only one controller, theres only one in "Device Manager" (Both "XBOX_360_DEVICE_01:03" in HIDs and "Xbox 360 Controller ..." in Xbox 360 Peripherals))?

masterx193, 2017-01-11:
same here, I guess you used standby or hibernation before? If you restart the problem is gone. Idk another fix

lamelizard, 2017-01-11:
That (sadly) shouldn't be it; I use neither hybernation nor standby. Also I had the Software installed for quite some time and had several restarts.

closed 2018-01-26 by nefarius