nefarius/ScpToolkit #371: Issue with game Spooky's Jumpscare Mansion

KRISTIANKULTA, 2016-07-08:
DualShock 3, came with PS3 Slim (Europe, Nordic)

When controller support is enabled in game settings, the controller keeps pressing W key. When I alt-tab out of game and click on any inputfield, it spams W until I close the game.

Jan200101, 2016-07-12:
Makes no sense because the controller is not emulating a keyboard its emulating a XBOX360 Controller so it isnt using WASD

KRISTIANKULTA, 2016-07-12:
It was apparently an old racing wheel behind sofa, that was plugged into my computer. The game does not use the DS3 controller, I try to press every single button but no response. I have enabled Controller Support from settings.

merklaw, 2016-07-21:
To: KRISTIANKULTA
Please test your controller in another game or application to be sure it is not the game who has a problem with using controllers.

KRISTIANKULTA, 2016-07-21:
Tested: Just Cause 2, Star Trek Online, War Thunder. Everything seems to be working.

That means Spooky's Jumpscare Mansion cannot use the controller. The game has controller support, there is setting to turn on that, but it could only get my racing wheel.

I did notice that War Thunder did not need ScpToolkit in order to use the controller, so I could test without Scp service's on and see if that does it.

closed 2016-07-21 by KRISTIANKULTA

HampDoesThings, 2017-09-02:
Please help! I want to use my Logitech F310 controller with the game but I cant stop spinning around. Please tell me how I can fix this!

(and yes controller support is enabled)