nefarius/ScpToolkit #372: Host address :  disconnected

mavalois, 2016-07-08:
Getting a problem here. Host address keeps showing as <disconnected> in SCP Monitor. SCP Bus Driver is running is the device manager. 

Needless to say, I'm unable to connect a PS3 controller

Logs attached, 

Anyone have a clue ?

[Logs.zip](https://github.com/nefarius/ScpToolkit/files/355199/Logs.zip)

awesomeness872, 2016-07-10:
do you have a bluetooth adapter? is it supported?

mavalois, 2016-07-11:
I do have a bluetooth adapter. I have not yet checked if it is supported. 

At the moment, I'm trying connect through USB before moving forward with anything else.

awesomeness872, 2016-07-11:
do the controllers work at all? I'm pretty sure that the host address is a bluetooth thing and shouldnt matter if you're using USB

mavalois, 2016-07-11:
Controllers are not working

maverick340, 2016-07-12:
I seem to have the same issue. Here are the steps I followed. 
1. Plugged in the DS3 controller (original)
2. Downloaded and installed SCPToolkit (no forced install)
3. Windows asked me if i trust these drivers source blah blah, I said Yes.
4. It shows all drivers installed fine.
5. SCP monitor still shows Host : Disconnected

Anything I am doing wrong here? It used to work with Windows 7 fine.

maverick340, 2016-07-12:
@mavalois  : Try the beta drivers : https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA
Worked for me.

mavalois, 2016-07-12:
Beta drivers suggested by maverick340 solved the problem.

closed 2016-07-12 by mavalois

rinkodijks, 2017-05-25:
Had the same problem. Beta drivers have solved the problem for me as well.

bucho2398, 2017-05-28:
Have the same problem, beta drivers solved nothing.

craazydude, 2017-11-05:
Have the same problem, beta drivers solved nothing.

lilcheo, 2017-12-05:
help drivers didnt work plz help

goldenmane, 2018-01-24:
Have the same problem, the controller connects through usb but there is no host adress in scp monitor, it says host adress disconnected even though I could connect my controller without any problems before the Windows 10 update.