nefarius/ScpToolkit #375: Application uninstaller doesn't uninstall drivers.

Ranzear, 2016-07-12:
Going back to O.G. ScarletCrush because there are too many other issues not worth DS4 support. Couldn't help but notice ScpToolkit left all its drivers installed when I removed it. I had to reinstall the toolkit, uninstall with the installer, then uninstall the toolkit again.

Is it so hard to find bluetooth dongles and controllers automatically? What if I don't have a cable for _every single controller_ I own to have them all plugged in at installation? Is this pop-up notifier used only for the installation and then left to rot in the system tray necessary?

Last straw was that the service would simply crash after a minute or two with no controllers connected, probably related to BT power modes if I had to guess, and would lose the host address _permanently_, even through restarts and driver reinstalls. I'm sure this is just specific to my BT dongle (loosest standard ever devised), but the screen-height error message wasn't helpful. If I could _find_ a log with the error in it I'd include it here.

nefarius, 2016-07-16:
Your input is highly appreciated.

merklaw, 2016-07-21:
To: Ranzear
Please understand that Scarlet.Crush is a project that has been active since at least 2011 and the ScpToolkit driver was only started in 2015. There is only one major developer (nefarius) and he is doing this project as a personal hobby, at least that is how I understand it.

You are welcome to submit patches to improve the functionality of the software.

nefarius, 2016-07-21:
@merklaw a 100% on the spot :smile:

closed 2018-01-26 by nefarius