nefarius/ScpToolkit #376: Motherboard bluetooth not visible v1.7

Pixolusion, 2016-07-12:
Love this toolkit

In your previous versions I was able to overwrite the motherboards bluetooth
In your latest wizard I think the filtering might be a little too aggressive as the device isnt listed anymore

Will fall back to your old versions for now

Keep up the good work!

nefarius, 2016-07-16:
I will consider adding an "advanced mode" to the installer which let's you do that again by hand. On the other hand; could you please provide the Hardware ID of your motherboard Bluetooth? Then I could add it to future releases as a supported device.

merklaw, 2016-07-21:
To: Pixolusion
You can find your Hardware ID of the Bluetooth adapter in Windows Device Manager if you right click on the device (found under "Bluetooth Radios") and click Properties, and then go to the Details tab and find "Hardware IDs". Please note if there are other entries that have something with "ID" in them, and post them here as well.

mirh, 2016-07-22:
#380

Pixolusion, 2016-08-20:
Qualcomm Atheros QCA61x4A Bluetooth 4.1

USB\VID_0B05&PID_1825&REV_0001
USB\VID_0B05&PID_1825

Sorry for the delay I havnt used this account for a bit
Thanks so much for looking into this

operamaniac, 2016-09-02:
I have the same issue too : 

Broadcom BCM43142A0 Bluetooth 4.0 + EDR
USB\VID_04CA&PID_2007&REV_0112
USB\VID_04CA&PID_2007

Hope this help. Keep up the good work!

closed 2018-01-26 by nefarius